frangoteam/FUXA #483: images not found

Virgil-N, 2022-09-14:
Resources is missing.
When I run command "ng serve" or "ng serve -c client" in client dir, I got this error:
http://localhost:4200/api/resources/images?type=images 404 (Not Found).

But if I run command "node main.js" in server dir, it's running well with no error.

mauroalexandre, 2022-09-14:
When you run node.main.js in server you are providing the static content and the another routes like imagesFileDir.

Look here https://github.com/frangoteam/FUXA/blob/8ab5a1edd0f6b65e7ce769d900f91e2bdc302c69/server/main.js#L252

And your root path "/" redirect to your static files https://github.com/frangoteam/FUXA/blob/8ab5a1edd0f6b65e7ce769d900f91e2bdc302c69/server/main.js#L243

When you run client separated, you not have the same routes avalaible in the same port for use root path /.

Virgil-N, 2022-09-15:
> When you run node.main.js in server you are providing the static content and the another routes like imagesFileDir.
>
> Look here [...]

I run client and server at the same time.
http://localhost:4200 not work.
http://localhost:1881 works well.
--------------
And one more question: what's the difference between them. Which one should I use and deploy?

mauroalexandre, 2022-09-15:
When you run server the dist folder (settings.httpStatic) is used to serve, so, if you need test client changes you need rebuild project.

Virgil-N, 2022-09-15:
> When you run server the dist folder (settings.httpStatic) is used to serve, so, if you need test client changes you need rebuild project.
[image]

Virgil-N, 2022-09-15:
[image]

Virgil-N, 2022-09-15:
Show u pictures.
[image]
[image]

One more thing, I only changed command to "ng build --base-href=/FUXA/ --output-hashing none" in package.json.
And what should I do to handle the error in picture 2 ?

mauroalexandre, 2022-09-15:
Sorry, maybe I'm not understanding you cenario.

But, I'll try explain again based in my idea.

When you run separated frontend you not have a imagePath in your route in
the same port.

When you run server, the server provide your dist in root path "/"./, and
another others routes described in main.js.

So, if you want use FUXA for develop maybe the best pratice is use server
for node (back) and front.

But for production is more recommended in my opinion you run separated npm
run for client and share another routes like imageData using webserver.

Virgil-N, 2022-09-15:
> Sorry, maybe I'm not understanding you cenario. But, I'll try explain again based in my idea. When you run separated frontend you not have a imagePath in your route in the same port. When you run [...]

Got it!
I'll use client just for show and use server for edit.

closed 2022-09-15 by Virgil-N