frangoteam/FUXA #484: Bind action with imported svg error

Virgil-N, 2022-09-14:
I upload a svg picture with the image button.
Then I set some actions, and when I switched to lab page, console show error below:

ERROR Error: ExpressionChangedAfterItHasBeenCheckedError: Expression has changed after it was checked. Previous value: 'ngForOf: '. Current value: 'ngForOf: c_71313341a1-813b,c_131202448a-b928,c_136154399-11065'.
    at viewDebugError (core.js:20342:1)
    at expressionChangedAfterItHasBeenCheckedError (core.js:20330:1)
    at checkBindingNoChanges (core.js:20432:1)
    at checkNoChangesNodeInline (core.js:23303:1)
    at checkNoChangesNode (core.js:23292:1)
    at debugCheckNoChangesNode (core.js:23896:1)
    at debugCheckDirectivesFn (core.js:23824:1)
    at Object.eval [as updateDirectives] (DataTableComponent.html:41:5)
    at Object.debugUpdateDirectives [as updateDirectives] (core.js:23813:1)
    at checkNoChangesView (core.js:23191:1)
View_DataTableComponent_0 @ DataTableComponent.html:41
proxyClass @ compiler.js:18234
push.../node_modules/_@angular_core@7.2.0@@angular/core/fesm5/core.js.DebugContext_.logError @ core.js:24041
push.../node_modules/_@angular_core@7.2.0@@angular/core/fesm5/core.js.ErrorHandler.handleError @ core.js:15762
(匿名) @ core.js:18028
push.../node_modules/_zone.js@0.8.29@zone.js/dist/zone.js.ZoneDelegate.invoke @ zone.js:391
push.../node_modules/_zone.js@0.8.29@zone.js/dist/zone.js.Zone.run @ zone.js:150
push.../node_modules/_@angular_core@7.2.0@@angular/core/fesm5/core.js.NgZone.runOutsideAngular @ core.js:17248
push.../node_modules/_@angular_core@7.2.0@@angular/core/fesm5/core.js.ApplicationRef.tick @ core.js:18028
(匿名) @ core.js:17912
push.../node_modules/_zone.js@0.8.29@zone.js/dist/zone.js.ZoneDelegate.invoke @ zone.js:391
onInvoke @ core.js:17289
push.../node_modules/_zone.js@0.8.29@zone.js/dist/zone.js.ZoneDelegate.invoke @ zone.js:390
push.../node_modules/_zone.js@0.8.29@zone.js/dist/zone.js.Zone.run @ zone.js:150
push.../node_modules/_@angular_core@7.2.0@@angular/core/fesm5/core.js.NgZone.run @ core.js:17203
next @ core.js:17912
schedulerFn @ core.js:13504
push.../node_modules/_rxjs@6.3.3@rxjs/_esm5/internal/Subscriber.js.SafeSubscriber.__tryOrUnsub @ Subscriber.js:196
push.../node_modules/_rxjs@6.3.3@rxjs/_esm5/internal/Subscriber.js.SafeSubscriber.next @ Subscriber.js:134
push.../node_modules/_rxjs@6.3.3@rxjs/_esm5/internal/Subscriber.js.Subscriber._next @ Subscriber.js:77
push.../node_modules/_rxjs@6.3.3@rxjs/_esm5/internal/Subscriber.js.Subscriber.next @ Subscriber.js:54
push.../node_modules/_rxjs@6.3.3@rxjs/_esm5/internal/Subject.js.Subject.next @ Subject.js:47
push.../node_modules/_@angular_core@7.2.0@@angular/core/fesm5/core.js.EventEmitter.emit @ core.js:13488
checkStable @ core.js:17258
onHasTask @ core.js:17302
push.../node_modules/_zone.js@0.8.29@zone.js/dist/zone.js.ZoneDelegate.hasTask @ zone.js:443
push.../node_modules/_zone.js@0.8.29@zone.js/dist/zone.js.ZoneDelegate._updateTaskCount @ zone.js:463
push.../node_modules/_zone.js@0.8.29@zone.js/dist/zone.js.Zone._updateTaskCount @ zone.js:291
push.../node_modules/_zone.js@0.8.29@zone.js/dist/zone.js.Zone.runTask @ zone.js:212
drainMicroTaskQueue @ zone.js:601
push.../node_modules/_zone.js@0.8.29@zone.js/dist/zone.js.ZoneTask.invokeTask @ zone.js:502
invokeTask @ zone.js:1744
globalZoneAwareCallback @ zone.js:1770
Show 3 more frames
DataTableComponent.html:41 
        
       ERROR CONTEXT DebugContext_ {view: {…}, nodeIndex: 15, nodeDef: {…}, elDef: {…}, elView: {…}}

unocelli, 2022-09-16:
Hi, thanks for the report.
I will fix it, however it should not compromise the operation

unocelli, 2023-10-11:
The issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2023-10-11 by unocelli