frangoteam/FUXA #494: Dialog of the input field stays open in editor mode

balincagabrielalin, 2022-09-16:
In my testing I managed to get into editor mode with a dialog of an input field open and it won't close, the only way to close that dialog was to refresh the page. 

Video :

https://user-images.githubusercontent.com/32384258/190588036-012b54cd-1ee5-4e3b-8078-dd53b5ca9c2a.mov

unocelli, 2022-09-16:
Hi, Thanks for the report.
which version of fuxa are you using?
do you have any errors on the browser console?

balincagabrielalin, 2022-09-16:
Hi!
The version I'm using is 1.1.8-833. And no there's no error in the browser console

unocelli, 2022-09-16:
let me understand, you have made changes to see the dialog also in editor mode? for which purpose?

balincagabrielalin, 2022-09-16:
No I haven't made any changes... It's a bug that I saw and reported it and I don't think this is how it should behave. I think that dialog box should close when the user click outside of it. I should add that when is in that state the dialog box become unresponsive(if isn't obvious from the video).

unocelli, 2022-09-16:
OK, I was able to reproduce it, thank you

closed 2022-10-12 by unocelli