frangoteam/FUXA #495: Is there a way to write MySql log?

tayfunonbasioglu, 2022-09-16:
Is there a way to write MySql log?

unocelli, 2022-09-29:
Hi,
what do you mean with log?

tayfunonbasioglu, 2022-09-30:
Hi,
In some projects, it may be necessary to save historical data to an existing mysql database. In this case, we can develop an easy and very useful method. I would like to share with you the method he developed for my own project.

[Video-Fri-Sep-30-2022-20-26-27.webm](https://user-images.githubusercontent.com/7642923/193324278-d5ab1cdc-8f10-41d6-80e4-8234ef601818.webm)

saufister, 2022-10-03:
> Hi, In some projects, it may be necessary to save historical data to an existing mysql database. In this case, we can develop an easy and very useful method. I would like to share with you the method he developed for my own project.
> 
>  Video-Fri-Sep-30-2022-20-26-27.webm

Please share step how to do that, thanks

closed 2024-09-20 by unocelli