frangoteam/FUXA #497: [Feature] Settings to auto resize view when window size is changed

mauroalexandre, 2022-09-17:
Hi!

In this PR #496 I created a settings to enable users choose if the view need auto resize if the size of window is changed, this feature is very appreciated for use the HMI in different display sizes.

Check demonstration in video below:

https://user-images.githubusercontent.com/68612845/190864680-90f9b30c-f3c9-4e75-b4e6-aba15765e08f.mp4

tayfunonbasioglu, 2022-09-17:
perfect, would you like to share with us

balincagabrielalin, 2022-09-19:
Hi!

I added your suggestion to my project to test it and it gives me this error in home.component.ts :

`ERROR in src/app/home/home.component.ts(344,45): error TS2551: Property 'Event' does not exist on type 'Window'. Did you mean 'event'?`

mauroalexandre, 2022-09-19:
I'll check. I belive is a conflict between DOM Event class name and
internal Fuxa name of class Event.

But, the feature worked?

balincagabrielalin, 2022-09-20:
Sorry for the delay, no it didn't worked because I couldn't compile the code.

unocelli, 2022-09-20:
Hi,
the feature look good but i think there is a problem with view size smaller than window size.

robsori, 2022-09-21:
Hi, it looks very good and I think it is a feature that complements the Fuxa. Can it be integrated?

unocelli, 2022-10-12:
I am going to close this as resolved. let me know if you have any issues.

closed 2022-10-12 by unocelli