frangoteam/FUXA #501: Cannot create device Tags from WebAPI Request

r0cheny, 2022-09-21:
Hi, 

I've been playing around with FUXA and I wanted to use my PostgreSQL DB in combination with postgREST as a Datasource, I was able to configure the Device and got my Request result: 
[image]

Now if i want to create a device tag, I can see the results but i cannot select anything as shown in the wiki. 
[image]

Is this intended? If so, what can I do to fix this issue? 
Thank you

unocelli, 2022-09-21:
Hi, you have an array then you have to define the key of tag id and the key of tag value
[image]

r0cheny, 2022-09-22:
Hi,
Thanks a lot, Now it works!

closed 2022-09-23 by unocelli