frangoteam/FUXA #506: BUG   -  InfluxDB   Creating duplicate records !

tayfunonbasioglu, 2022-09-23:
When a new record information comes, it saves again with the previous record. Therefore, 4 records are created for 2 Tags in the same time period.

[image]

unocelli, 2022-09-24:
Hi, thanks for the report.
it would be useful to know if the same problem occurs with the same configuration but on sqlite.

tayfunonbasioglu, 2022-09-24:
Hi ,   sqlite no problem.

[image]

unocelli, 2022-09-25:
Hi, I suppose you have a polling interval smaller than 1 second and the reason is that timestamp in influxdb is truncate in seconds (you see the number of digits in sqlite are 13).
I checked and I see that the value are write in db after a conversion from milliseconds to Date...

function writePoint(tag, deviceName) {
  const point = new Point(tag.id)
  .tag('name', tag.name)
  .tag('type', tag.type)
  .timestamp(new Date(tag.timestamp));
  writeApi.writePoint(point);
}

haitan981, 2022-11-25:
How do you query the data? I open the .db file and it's empty
[image]

unocelli, 2023-02-05:
@haitan981 you have to enable the storage of tags data

closed 2023-02-05 by unocelli