frangoteam/FUXA #513: Can't connet to MODBUS-RTU.

minicoco, 2022-09-28:
Hello everyone. I want use FUXA to connect MODBUD-RTU device, this is my config:
[image]

I use KEPServerEX create a modbus-rtu slave：
[image]

BUT they can't connet successfully:
[image]

I use other tools ,like "Mthings" ，they can communicate successfully:
VSPD look like :
[image]

Sorry my poor english,I hope I made it clear。

Did I do something wrong? Please give me some suggestions. Thank you !

unocelli, 2022-09-29:
Hi,
have you try to enable 'Fragmented' option?

afbenevides, 2022-10-08:
Hello, has this issue been resolved? We have similar issue, can’t connect to modbus RTU on pi. Modbus works if out of fuxa, but cant connect through FUXA. Any clue?

afbenevides, 2022-10-08:
Other info, we use /dev/tryusb0 as port address, is that right ? Or do we need to use another form to point to modbus?

unocelli, 2022-10-14:
Hi, have you try to remove from FUXA plugins and install the driver from server folder?
`npm install modbus-serial`

afbenevides, 2022-10-21:
Forgot to comment but installing Modbus-serial from npm fixed the issue for me. Thanks

unocelli, 2022-11-13:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2022-11-13 by unocelli