frangoteam/FUXA #515: How to fix error: Cannot read properties of undefined (reading 'retention')

haitan981, 2022-09-29:
After I follow the instructions as below:
Or [Download the latest release](https://github.com/frangoteam/FUXA/releases) and unpack it (second option)

cd ./server
npm install
npm start

 The screen shows the following error message:

D:\FUXA\FUXA-1.1.8\FUXA-1.1.8\server>npm start

> fuxa-server@1.1.8-821 start
> node main.js

2022-09-29T08:26:51.252Z [INF]  FUXA V.1.1.8-821
2022-09-29T08:26:51.906Z [INF]  FUXA init in  716ms.
2022-09-29T08:26:51.913Z [INF]  FUXA started!
2022-09-29T08:26:51.916Z [INF]  WebServer is running http://127.0.0.1:1881/
2022-09-29T08:26:53.076Z [INF]  socket.io client connected
**2022-09-29T08:27:06.939Z [ERR]  TypeError: Cannot read properties of undefined (rea**ding '**retention')**
    at D:\FUXA\FUXA-1.1.8\FUXA-1.1.8\server\runtime\storage\daqstorage.js:107:74
    at new Promise (<anonymous>)
    at Object.checkRetention (D:\FUXA\FUXA-1.1.8\FUXA-1.1.8\server\runtime\storage\daqstorage.js:105:12)
    at _clearStorage (D:\FUXA\FUXA-1.1.8\FUXA-1.1.8\server\runtime\jobs\cleaner.js:50:35)
    at D:\FUXA\FUXA-1.1.8\FUXA-1.1.8\server\runtime\jobs\cleaner.js:24:27
    at new Promise (<anonymous>)
    at Cleaner.execute (D:\FUXA\FUXA-1.1.8\FUXA-1.1.8\server\runtime\jobs\cleaner.js:19:16)
    at D:\FUXA\FUXA-1.1.8\FUXA-1.1.8\server\runtime\jobs\index.js:168:61
    at Array.forEach (<anonymous>)
    at D:\FUXA\FUXA-1.1.8\FUXA-1.1.8\server\runtime\jobs\index.js:166:26
    at new Promise (<anonymous>)
    at _checkJobs (D:\FUXA\FUXA-1.1.8\FUXA-1.1.8\server\runtime\jobs\index.js:163:16)
    at _checkStatus (D:\FUXA\FUXA-1.1.8\FUXA-1.1.8\server\runtime\jobs\index.js:98:25)
    at Timeout._onTimeout (D:\FUXA\FUXA-1.1.8\FUXA-1.1.8\server\runtime\jobs\index.js:31:17)
    at listOnTimeout (node:internal/timers:564:17)
    at process.processTimers (node:internal/timers:507:7)

How to fix that error to start FUXA. Please help me. Thank you

unocelli, 2022-09-29:
Hi, Thanks for the report. the issue is already fixed #474 where you find the a workaround

malcore10, 2022-10-11:
nodejs, npm, os version?
try to remove and install it again

haitan981, 2022-10-11:
> nodejs, npm, os version? try to remove and install it again

Nodejs V18.9
How to remove Fuxa and reinstall it

mauroalexandre, 2022-10-11:
> > nodejs, npm, os version? try to remove and install it again
> 
> Nodejs V18.9 How to remove Fuxa and reinstall it

Just git pull from master if you want to use the latest version or choose one tag.

Remove your node_modules folder and run npm install.

malcore10, 2022-10-11:
But first remove the nodejs version and install V14 please!