frangoteam/FUXA #516: History charts can't draw curves

artofegg, 2022-09-29:
[image]
Just finished setting it up, the charts worked fine, and after about a few hours, it looked the way it does now.

unocelli, 2022-09-29:
Hi, Thanks for the report. It looks like a refresh problem of the control.
Can you make some test, so that I can reproduce it? Browser used? Click to refresh chart? Browser refresh F5? has that ever worked?

artofegg, 2022-09-29:
Browser is firefox  105.0.1 64bit for Win 10 , i also tested at the last Chrome for all recommended actions. 
Ten hours later , looks part of it is working up , and first charts can display by local zoom operation use mouse

[image]

saufister, 2022-10-02:
Its also happen to my chart, actually i am from vue js doesnt understand to fix angular js project☺🙏

unocelli, 2022-10-25:
Hi,
I think I have been able to reproduce it and I think the problem is that the FUXA server has recorded 'undefined' values. I will try to fix the issue so that only valid values are recorded.

artofegg, 2022-10-26:
I guess the problem is the mqtt server disconnect randomly , NaN data received by database and web drawing failed. Which area do not have NaN data so it can be selected with the mouse and succeed.

artofegg, 2022-10-27:
> Hi, I think I have been able to reproduce it and I think the problem is that the FUXA server has recorded 'undefined' values. I will try to fix the issue so that only valid values are recorded.

Great minds think alike, thanks your job !

unocelli, 2022-11-13:
I am going to close this as resolved. let me know if you have any issues.

closed 2022-11-13 by unocelli