frangoteam/FUXA #517: Auto run script

haitan981, 2022-09-30:
Hi everyone
How to make script run automatically without Click or Release Event

mauroalexandre, 2022-09-30:
Use "Script schedule".

[image]

haitan981, 2022-09-30:
Thank you. It worked fine

closed 2022-10-12 by unocelli

Jue07, 2024-03-21:
I am using ver.1.1.7 of Fuxa. Which one should I choose to run the script automatically?

1.Loop with interval
2. Only on start
3. Scheduling

unocelli, 2024-03-26:
Hi, interval at 0 seconds is not an interval...disabled!

Jue07, 2024-03-27:
Thanks for the tip!

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>