frangoteam/FUXA #518: Cannot start FUXA via cmd

haitan981, 2022-09-30:
Hi everyone
I start FUXA via command but got error like below image. If anyone knows how to fix it, please help me. Thank so much
[image]

unocelli, 2022-09-30:
Hi, The issue is already fixed https://github.com/frangoteam/FUXA/issues/474 where you find the a workaround

unocelli, 2022-10-12:
I assume that in the meantime you have installed fuxa by following the instructions on the [wiki](https://github.com/frangoteam/FUXA/wiki/Installing-and-Running). let me know if you have any issues.

closed 2022-10-12 by unocelli

haitan981, 2022-10-13:
> I assume that in the meantime you have installed fuxa by following the instructions on the [wiki](https://github.com/frangoteam/FUXA/wiki/Installing-and-Running). let me know if you have any issues.

I have reinstalled Node JS v14 and new FUXA 1.1.9 but still not working. How to fix that error.

[image]

malcore10, 2022-10-13:
please a screeshot:
node -v & npm -v

haitan981, 2022-10-14:
> please a screeshot: node -v & npm -v
I did as per your request
[image]