frangoteam/FUXA #520: Alarms Table

tayfunonbasioglu, 2022-10-01:
Is it possible to save the alarms table to influxDB instantly?

[image]

saufister, 2022-10-02:
Hei, what kind graph that your used ?

tayfunonbasioglu, 2022-10-02:
> Hei, what kind graph that your used ?

hi, external php file, highcharts.. 

[https://www.highcharts.com/demo](url)

[image]

[image]

saufister, 2022-10-02:
Wow that so cool, so are u using iframe to show that graph? , how about loading time with iframe ?,  i plan to build graph using vue js and show with iframe to, with stream data from mongodb😃

tayfunonbasioglu, 2022-10-02:
> 

> Wow that so cool, so are u using iframe to show that graph? , how about loading time with iframe ?, i plan to build graph using vue js and show with iframe to, with stream data from mongodb😃

I'm using mysql, somehow I managed to register Mysql with Fuxa. I can create charts and queries with php and highchart on localhost. For iframe refresh it is simple but effective ;)

<meta charset="utf-8">  
<title>ZeroScada</title>  
<meta name="viewport" content="width=device-width,initial-scale=1">  
<meta http-equiv="refresh" **content="10"** >

tayfunonbasioglu, 2022-10-02:
> Wow that so cool, so are u using iframe to show that graph? , how about loading time with iframe ?, i plan to build graph using vue js and show with iframe to, with stream data from mongodb😃

I'm using mysql, somehow I managed to register Mysql with Fuxa. I can create charts and queries with php and highchart on localhost. For iframe refresh it is simple but effective ;) There is no iframe refresh feature in fuxa for now.
`<meta http-equiv="refresh" content="10" >` --> 10 second refresh page

haitan981, 2022-10-10:
> > Hei, what kind graph that your used ?
> 
> hi, external php file, highcharts..
> 
> [https://www.highcharts.com/demo](url)
> 
> [image]
> 
> [image]

How to you can insert a photo (Zerosystem) in to header

martinjug, 2026-02-12:
Saving alarms in the same database as process data would be a great feature. For example, saving alarms as well in InfluxDB would enable querying both process data and alarms together to build dashboards and reports for higher-level systems.