frangoteam/FUXA #521: Why your screengrabs so damn fast?

dustymac, 2022-10-02:
Whoever thought that showing us blazing fast screengrab walkthroughs for your FAQ page needs to be fired.  Immediately.  I will gladly help you re-record each of them if you will at least admit they are abnormally too fast for the average Joe.  And perhaps even too fast for the average Schmo.

unocelli, 2022-10-06:
?

closed 2022-10-06 by unocelli

dustymac, 2022-10-06:
Apologies -- my frustration got the best of me, and resulted in a poorly worded issue on Github.  I actually ended up downloading your .gif WIKI videos and slowed them down at https://ezgif.com/speed.

Your WIKI is great.  Very helpful for me learning my way around FUXA.  But the speed of the .gif files is quite fast for trying to learn.

Like this one:

[image]

mauroalexandre, 2022-10-06:
This is an open source project, including wiki. Please if you can contribute, do it, before creating poor issues.