frangoteam/FUXA #524: 'MODBUSTCPIP' _readMemory error! Error: Modbus exception 1: Illegal function (device does not support this read/write function)

haitan981, 2022-10-03:
The system is working normally, suddenly an error like the screenshot below appears, and the system stops working and cannot be restarted. Anyone know the cause, help me fix the error.
Thank you so much

[image]
[image]
[image]

unocelli, 2022-10-30:
Hi, try to remove from FUXA plugins and install the driver from server folder.
`npm install modbus-serial`

unocelli, 2023-02-05:
This issue should be fixed in the latest version. let me know if you have any issues.
In master branch you find a version where the library will be installed as standard.

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli