frangoteam/FUXA #526: How to fix Cannot start running FUXA

haitan981, 2022-10-04:
The system is working normally, an error appears automatically as shown in the screenshot below, and the system stops working and cannot be restarted. Who knows the staff, help me fix the error.
 Error: ENOENT: no such file or directory, scandir
[image]
[image]

malcore10, 2022-10-05:
Hi,
check nodejs version. (supported v10-14)

haitan981, 2022-10-06:
> Hi, check nodejs version. (supported v10-14)

I am using Node JS 18.9.0 . new latest

malcore10, 2022-10-06:
now you know why it doesn't work. probably the other issues #524 #518 are also for the same reason

haitan981, 2022-10-07:
> now you know why it doesn't work. probably the other issues #524 #518 are also for the same reason

I don't see any similarity in those errors

haitan981, 2022-10-11:
> now you know why it doesn't work. probably the other issues #524 #518 are also for the same reason
This morning I started FUXA again this error appeared on the cmd screen.
How to fix that error ? Please help me . Thank you so much

[image]

unocelli, 2022-10-12:
I assume that in the meantime you have installed fuxa by following the instructions on the [wiki](https://github.com/frangoteam/FUXA/wiki/Installing-and-Running). let me know if you have any issues.

closed 2022-10-12 by unocelli