frangoteam/FUXA #531: Real digit format from Siemens S7

gosztek, 2022-10-10:
Hi, I have a problem - I do not see the option to set the number of decimal places in the display of values. Where should I look?
[image]

[image]

unocelli, 2022-10-10:
Hi, you should see like this:
[image]
which version are you using?

haitan981, 2022-10-11:
> Hi, you should see like this: [image] which version are you using?

what version are you using . I am using v1.1.8-821 ,so it do not have Format digits

gosztek, 2022-10-11:
> Hi, you should see like this: [image] which version are you using?

Many thanks for the prompt reply. I have 1.1.8-821 version.

haitan981, 2022-10-11:
> > Hi, you should see like this: [image] which version are you using?
> 
> Many thanks for the prompt reply. I have 1.1.8-821 version.
I also use the same version as yours. Why don't I have .You have up to 3 input boxes
[image]

haitan981, 2022-10-11:
> > Hi, you should see like this: [image] which version are you using?
> 
> what version are you using . I am using v1.1.8-821 ,so it do not have Format digits

gosztek, 2022-10-18:
Do we know anything about displaying these float variables? Any additional ideas?

unocelli, 2022-10-18:
to be sure I would install the last version 1.1.9

gosztek, 2022-10-21:
FUXA works fine after upgrading to version 1.1.9. Regards
[image]

closed 2022-10-21 by unocelli