frangoteam/FUXA #532: Modbus connection randomly hangs after modbus slave reconnect or network instability

algismo, 2022-10-10:
Hi,

I noticed that modbus connection(s) between FUXA server and slaves may randomly hang after a slave reconnect or network instability.

When this issue happens, then:
1. FUXA client indicates connection loss with the slave (this toast notification/popup is appearing constantly)
2. Slave is online but wireshark does not indicate any attempt to make a new connection from the FUXA server side
3. Nothing is being printed to the the FUXA server output
5. The only way to fix this issue is to restart FUXA server.

I suspect that this happens when re-connection of the slave happens quicker than disconnect is detected in FUXA server (indicated by these "... working (connection || polling) overload! X").

I use version v1.1.7.

Has anybody observed anything similar?

Algirdas

algismo, 2022-10-13:
I managed to reproduce the issue in the test environment with a network simulator, by introducing random resets of a Modbus TCP connection.

The result is:
1. FUXA server device state machine thinks that connection with the modbus device is lost. I see this log message beig printed once `2022-10-13T10:21:02.866Z [ERR]  '..dev name...' connect failed! Error: TCP Connection Timed Out`
2. Then server tries to reconnect by calling method `connect` (/FUXA/server/runtime/devices/modbus/index.js, line 38)
3. On line 45 in the `connect` method there is a check if internal modbus client is connected or not:
`if (!client.isOpen  && _checkWorking(true))`. Here `client.isOpen` actually returns true (internal client is connected) and connect method stops by calling reject() method (that leads nowhere) and emitting status `_emitStatus('connect-error');`
5. Now the server is in a deadlock and it will never recover from this state, device is not polled, tags are not updated.
6. The fix that works is to close internal client when this situation occurs. It worked when I did a quick test.
7. The questions that I still have no answer are:
-- Why this "miss-synchronization" of states happens.
-- If closing internal client on connect is a good overall solution. More investigations are needed here.

@unocelli, any comments/ideas from your side?

unocelli, 2022-10-14:
Hi, have you try to close connection before open?

```
this.connect = function () {
        return new Promise(async (resolve, reject) => {
            if (data.property && data.property.address && (type === ModbusTypes.TCP || 
                    (type === ModbusTypes.RTU && data.property.baudrate && data.property.databits && data.property.stopbits && data.property.parity))) {
                try {
                    await this.disconnect();
                    if (!client.isOpen  && _checkWorking(true)) {
```

algismo, 2022-10-16:
Hi, 

I did almost the same with the code below and it works ok so far:

```
diff --git a/server/runtime/devices/modbus/index.js b/server/runtime/devices/modbus/index.js
index 7c20995..7f471c4 100644
--- a/server/runtime/devices/modbus/index.js
+++ b/server/runtime/devices/modbus/index.js
@@ -62,6 +62,10 @@ function MODBUSclient(_data, _logger, _events) {
                             _checkWorking(false);
                         });
                     } else {
+                        if(client.isOpen){
+                            logger.error(`'${data.name}' connection is not closed properly. Cleaning up.`);
+                            client.close();
+                        }                        
                         reject();
                         _emitStatus('connect-error');
                     }
``` 

I do print an error when this happens and close the internal modbus client...
I have a quick solution for now for my needs. Will test your proposal later and will make a PR for one or another solution.

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli