frangoteam/FUXA #543: How to change the logo and tittle wint V1.1.9

15961773577, 2022-10-14:
Hello,FUXA team
I was used command ( npm install -g --unsafe-perm @frangoteam/fuxa
fuxa)
to install the fuxa. And i aslo saw the issue #491,but i still wonder with these,because in my armlinux system i can't find this files.
How can i change this in my system ?

haitan981, 2022-10-14:
If you install it this way, you will have the file you want
[Download the latest release](https://github.com/frangoteam/FUXA/releases) and unpack it (second option)

cd ./server
npm install
npm start

QCPaul1920, 2022-10-14:
> 

thank you for your reply
I have already installed and started the fuxa , i'm just want to change the logo image now . But can't find the path to change this with issue #491 .

haitan981, 2022-10-14:
> > 
> 
> thank you for your reply I have already installed and started the fuxa , i'm just want to change the logo image now . But can't find the path to change this with issue #491 .

It is in #490

afbenevides, 2022-10-22:
Hello, I would also like to use the logo feature in layout/Navigation side menu/drop down named logo.

I just wonder where we need to put the logo so it shows in that drop down? Also what type of file is accepted(png, svg, ...) ?

Thanks for the help.

mauroalexandre, 2022-10-22:
> Hello, I would also like to use the logo feature in layout/Navigation side menu/drop down named logo.
> 
> I just wonder where we need to put the logo so it shows in that drop down? Also what type of file is accepted(png, svg, ...) ?
> 
> Thanks for the help.

Look [here](https://github.com/frangoteam/FUXA/blob/aebd0494908c0dc544966128c2109e278f80d763/server/api/resources/index.js#L49) 

Allowed extensions is ['.jpg','.jpeg', '.png', '.gif', '.svg']

afbenevides, 2022-10-22:
Thanks @mauroalexandre for trying to help me. However, I still can't find the path where to insert new logos so I can choose them in the drop down menu.

If it helps I also noted that even if I see FUXA in the side bar as logo, no Fuxa is available in the dropdown menu. In fact the drop down menu is empty whatever I do.

I tried adding a .png file in here without success: (also tried rebooting to reload if necessary...)

/usr/local/lib/node_modules/@frangoteam/fuxa/_images
/usr/local/lib/node_modules/@frangoteam/fuxa/api/resources
/usr/local/lib/node_modules/@frangoteam/fuxa/api/resources/images

Im on a PI, with version 1.1.9-883
installed with:
sudo npm install -g --unsafe-perm @frangoteam/fuxa

unocelli, 2022-10-23:
Hi,
try to add in a sub folder, example `/usr/local/lib/node_modules/@frangoteam/fuxa/_images/logos`

QCPaul1920, 2022-10-24:
> 

Thank you unocelli,
I add an image by you suggestion,now it's looks perfect!!

closed 2022-10-25 by QCPaul1920

jinyistudio534, 2022-12-28:
> same here thanks, it now works
I have create **/usr/local/lib/node_modules/@frangoteam/fuxa/_images/logos** but I cann't see any file too.
[image]

unocelli, 2022-12-28:
You have created the full path or only logos folder?

jinyistudio534, 2022-12-28:
> You have created the full path or only logos folder?

full path !
[image]