frangoteam/FUXA #546: Input control. Multiple issues after latest changes?

algismo, 2022-10-16:
Hi,

I switched to the latest revision (master) and noticed some regression issues with the **Input** control:

1. Units/Format are not visible in configuration anymore. Are you aware? Somebody is working on this?
2. Unpolished behavior when update is disable, user enters value outside of a range and then clicks somewhere in a view or presses ESC. It is not possible to "Cancel" editing kind of.

unocelli, 2022-10-16:
Hi, I'm sorry, I think that in making some changes I forgot to check unit a format, I will try to fix that as soon as possible.

algismo, 2022-10-17:
No problem, I like the addition of the new option "Only Number". It makes things cleaner.

Quite minimal changes are needed to get back the units and format. There is more of a problem with the Input type "number" that you set on the HTML control when "Only Number" is enabled. I was testing the same approach in the beginning, but the downside is that then "unit" text cannot be appended, as control only accepts numbers. What are you planning to do here?

unocelli, 2022-10-26:
Hi, sorry for the delay.
You are right...I will try to overlap a label.