frangoteam/FUXA #550: Retain Information after restart

lakii22, 2022-10-18:


unocelli, 2022-10-18:
Hi, to see the history you can use the chart control or the table, but it would not be correct to show a value from the history where you would expect the actual one

lakii22, 2022-10-19:
Hi @unocelli, thank you for your quick answer, but I think that I've not expressed myself well in the comment above.

What I've tried to mean is, that I'd like to have the last "valid" value from a variable after a system restart (if the value haven't changed of course), instead of 0 or "empty".

But reading what you've wroted above I can imagine that this isn't possible right now because of the DAQ format.

Just to give you a tip of why I'm trying to "retain a value from a variable": My goal is to make a "setup page" where I can set the amount of machines present in the factory and some other informations about quantity that I can use to dinamically set the view size, layout, machine names and so on. 

So right now I think that the only option that I've is to use a SQL DB through WEBapi GET or to hardcode the value that I want into a script, right ? So I could keep the value even after a restart from the server.

unocelli, 2022-10-26:
eventually you can try to save the values to a file with a script

```
const fs = require('fs');
fs.writeFileSync(filename, values);
```

unocelli, 2022-11-13:
I am going to close this as resolved. let me know if you have any issues.

closed 2022-11-13 by unocelli