frangoteam/FUXA #551: Internal variables - error "Plugin is missing"

lakii22, 2022-10-18:
Creating a connection of the type "internal" is returning me the following warning:
[image]

After creating the connection I can still add the variables, however if I try to change the values nothing happens.
[image]

Am I making a mistake or this is actually how it should work ?

unocelli, 2022-10-18:
Hi, Thanks for the report.
It is a warning not a error. I will fix...in any case has no consequences on the functionality. 
internal device is used for frontend (client) logic, if you are using scripts you should use FUXA (server) or other device types (not internal)

lakii22, 2022-10-19:
@unocelli Thank you again for the feedback. For me is clear... 

I've made a mistake about the function of these variables (even with the quote "Frontend only" in the connection creator 😅).

So I'm going to close this Issue.

closed 2022-10-19 by lakii22