frangoteam/FUXA #561: Historic Trend is not shown

sodaGin, 2022-10-25:
First of all, very nice project!

I have set up a system with several SiemensS7_PLCs with an Live Line-Chart and a Historic Line-Chart.
For Historic Linechart I use the same variable like the live chart.
The variables are "Registration Enabled".

Somehow the historic trend is not working and not showing any values.
What I get is following Warning:
[image]

When I set everything up with only one connection, I get the same warning.
I can´t find the Problem. What am I doing wrong?
Thx
sodaGin

unocelli, 2022-10-26:
This happens when the polling interval is shorter than the time it takes the server to read and handle the tag values from the PLC.
How many PLC and Tags are you try to connect? HW? OS?

gosztek, 2022-11-02:
It is similar for me - no data on historical charts.

[image]

Data collection configured correctly. Any ideas?

[image]

unocelli, 2023-02-05:
This issue should be fixed in the latest version. let me know if you have any issues.

closed 2023-03-11 by sodaGin