frangoteam/FUXA #574: How to import a library in a script

mariorl, 2022-11-03:
Hi, I'm looking for a way of save several user values on a local db, for example sqlite. So I'm using the script feature. But if I try to import modules with 'require' on the script, it wont work for me, and server log shows a message like "sqlite3 module not found". Since I noticed the script are not .js files but db items in the project database, is there a way to include an external module? Thanks!

unocelli, 2022-11-04:
Hi, I think the best solution is to add your code directly into the server, without use the script editor in fuxa.

mariorl, 2022-11-04:
ok, but js is not a skill on me, i think it is better to leave the server quiet. Fortunately I foud another issue where you answered to another user about to use 'fs' module, (¿filesystem?) and it worked for me also, within script editor. I used it archiving small contents like iframe server config, so that I serve flask content inside fuxa iframe, and python reads the config form this config file.

unocelli, 2022-11-13:
I am going to close this as resolved. let me know if you have any issues.

closed 2022-11-13 by unocelli