frangoteam/FUXA #576: OBJECT COPY AND PASTE DOES NOT COPY OBJECT ANIMATIONS - EVENTS

mariorl, 2022-11-04:
Hey there, I get shocked when noticed, some time after testing Fuxa, so that when I animate an object, in "Shape settings/property pane" and set seral fill/stroke color for an item, the animation works gret, but when I try to copy an paste the whole configured item in designer view, lets say, a copy a motor or valve, the animations are not copied to the new object. May be should temporaly solve it by editing the project db with an external sqlite db browser.

unocelli, 2022-11-04:
Hi, please let us know the version that are you using and the shape. I cannot reproduce it.

mariorl, 2022-11-04:
Fuxa 1.1.9 master. I went ahead testing, and noted the issue not happends with all objects with animation capacity. For example a simple square shape linked to a PLC var copy and paste OK along with all animations. The ones who not worked are text and resource/svg. These ones copy and paste without "animation" properties for me. Also noted, if I export a view and then import back, the animations are saved and retrieved ok in the new view. If I edit in external json editor the missing nodes for example for a motor svg animation (  missed "ranges"[] and "variableid":   ) it get fixed when imported to fuxa editor.

unocelli, 2022-11-04:
Ok, we have to fix it, thanks.
about the svg with animation: currently fuxa editor is inflexible and remove the elements that not support. can you send me the svg file?

mariorl, 2022-11-05:
I noticed in the corresponding fuxa project sqlite table to that the view/svg belongs, who the missed arguments on copied element are "ranges"[] and "variableid": filled with the same values of the original view/svg element. Filling them with external text tool fixes it.
Also, I just tested again, for me a text inserted with "text tool", when edited color of text depending on a PLC variable, if then you copy an paste that text in the view designer in order to obtain several texts with same properties, is copied without animations.
[image]

mariorl, 2022-11-14:
Hi, I noticed that, rather than raw copy an paste objects in the editor, the button "duplicate" works OK for copy simple object and its animations, like texts, rectangles etc. So the issue sould be focused only to "Resource" objects, like the pump in previous post. Thanks.

unocelli, 2023-02-05:
I am going to close this as resolved. let me know if you have any issues.

closed 2023-02-05 by unocelli