frangoteam/FUXA #577: Can't Write to opcua tag using a button

henjoe, 2022-11-05:
I am trying to write "100" value on my simulation opcua server. 
However, it doesn't write well, here is my configuration on my button.

[image]

unocelli, 2022-11-05:
Hi, what type is it?

henjoe, 2022-11-05:
Hi @unocelli , 

It is a float (number) type of data.

unocelli, 2022-11-05:
Can you send me a screenshot of device settings?
[image]

henjoe, 2022-11-05:
Hi, 

fyr.

[image]

unocelli, 2022-11-06:
Can you send me the log file, or the output error in server? can you change the value of the other Counter?

unocelli, 2023-05-14:
I will therefore close this issue, feel free to re-open it if something is still not working.

closed 2023-05-14 by unocelli