frangoteam/FUXA #591: Some times  the items in DateRange select options have not been translated

GitTurboy, 2022-11-14:
shown like the image below
[image]

GitTurboy, 2022-11-15:
I found the reason of this problem:
At line 733 of  [client/src/app/gauges/gauges.component.ts 
       this.translateService.get(chartRange[key]).subscribe((txt: string) => { chartRange[key] = txt; });
we subscribe the response from translateService, if this function take more time than normal , return the result before the chart rendered, the select option items will keep untranslated status.
@unocelli ,do you think this thougth is right?

unocelli, 2022-11-15:
Hi, Thanks for the report. 
Yes, I think you are right, I can reproduce it. I have to fix it.

unocelli, 2023-02-05:
I am going to close this as resolved. let me know if you have any issues.

closed 2023-02-05 by unocelli