frangoteam/FUXA #595: Why some bar graph's settings are doubled?

IonuDragomir1, 2022-11-15:
Hello, 

Could you please tell me why some bar graph's settings (Title, Value Axis, Category Axis, Legend) are doubled? 
If I change, for example, title font size, this is done in both Title Font size fields.
[image]

unocelli, 2022-11-15:
Hi,
I can't reproduce it and if you want to get a response from reporting issue, you need to take the time to investigate and provide adequate information how we can reproduce it.

please version of: fuxa, os, node, npm, browser etc.

IonuDragomir1, 2022-11-18:
Sorry, it was my mistake. I solved the issue. Thanks!

closed 2022-11-25 by unocelli