frangoteam/FUXA #597: Is there any bulk import functions?

gahang, 2022-11-17:
Hi, I have thousands of OPC-UA nodes to add in and can only add them one by one in the web page, that's a big work to do. Can I add them by editing the configuration files/database or having a bulk import function or something like that?

malcore10, 2022-11-17:
you can export and import the devices in json and csv format
[image]

gahang, 2022-11-18:
Thanks for the answer. How can I configure the alert point like this?

unocelli, 2022-11-25:
currently fuxa does not have import/export functions for alarms

gahang, 2022-11-28:
Can I add alarms by editing the configuration file or database?

unocelli, 2022-12-01:
You can edit the project exported in json, not really comfortably or you can edit the table `alarms` in sqlite database `server/_appdata/project.fuxap.db`