frangoteam/FUXA #607: Discord community for FUXA?

Rafael-OliveiraBR, 2022-11-24:
Hey guys.

A lot of open source communities have an discord channel for members to interact to each other (voron, printNC, flipper, for example).
Since this is a under development project, i think would be good for users to have a channel to interact with others without the need for open issues here.
Keeping people close bring new ideas to the table, makes feedback more reliable, can become a source for refference materials, people often try to help with issues from others. 
Having a discord channel could make FUXA evolve faster.
Just an idea.

: )

unocelli, 2022-11-25:
Hi, thanks for the suggestion, really interesting. totally new to me, i just opened a channel (i think) can you give me some tips
https://discord.gg/Gp8H4w6Pmh

unocelli, 2023-02-05:
Now available

closed 2023-02-05 by unocelli