frangoteam/FUXA #608: Cannot send email by SMTP

haitan981, 2022-11-24:
I want to use the SMTP feature to send daily reports via mail, but when I press Test, the screen shows an error message like below.
How to fix that error ? Thank everyone.
[image]

unocelli, 2022-11-25:
Hi, gmail doesn't work, I think they have disabled access via smtp, you must have another server that allows access via smtp, for example https://sendgrid.com/ and use apikey

unocelli, 2023-02-05:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2023-02-05 by unocelli

jay-jjwu, 2023-07-28:
@unocelli ,hi uncoil, i got the same issue, i did test the email server with another application, and it works well. but it seems always getting an error "greeting never received" on fuxa sys log. 
BTW, does the system support SSL login to the mail server? Because my mail server only support SSL connection becuase of security. Is that could be the possible reason for fail?

unocelli, 2023-08-06:
Hi, which port are you using?
```
    const transporter = nodemailer.createTransport({
        host: smtpServer.host,
        port: smtpServer.port,
        secure: (smtpServer.port === 465) ? true : false, // true for SSL
        auth: {
            user: smtpServer.username,
            pass: smtpServer.password
        }
    });
```

jay-jjwu, 2023-08-15:
i'm using 465, the error logs shows as below.

 [error] api post sendmail: Connection closed unexpectedly
[image]

robbudge, 2024-08-22:
@unocelli, I am also having an SMTP SSL Greeting never received via port 465 / SSL.
My email server is SSL only.
I am testing Version 1.2.0-1670 on a Linux host.
I have verified the host machine and email setting work correctly by configuring a local client on the Linux box and all is good,
I was initially suspecting a Firewall or DNS issue.

If I change email providers to SendGrid and use 587 (Non SSL) all is good.
But that flags the email as spam. When ever I use port 465 with my correct server I get greeting never received ????

If you DM me i can provide you with Server information if needed