frangoteam/FUXA #609: Bar Graph - Color by value

AlexandruAndrei3, 2022-11-24:
Hi, it would be a nice feature if you could add a feature sou we can add bar graph colors like in image shown bellow. I saw that this on another objects alredy exists.
[image]

unocelli, 2022-11-25:
Hi, currently is not planned, but you could use several bar gauge in line and each has its own range.
[image]

AlexandruAndrei3, 2022-11-25:
Thanks i will keep in mind your suggestion but i would like that the color of the entire bar graph to be changed.

unocelli, 2024-08-01:
Hi, #1285 with more options

closed 2024-08-01 by unocelli

DHIAFIQ, 2024-08-30:
> Hi, it would be a nice feature if you could add a feature sou we can add bar graph colors like in image shown bellow. I saw that this on another objects alredy exists. ![BarGauge](https://user-images.githubusercontent.com/94168341/203817694-21789c9f-7b90-49ba-bd5a-65cbf58c6869.JPG

why in my fuxa not have option to add another min and max