frangoteam/FUXA #610: Modbus serial not work

2345lug, 2022-11-24:
Hi! I created new modbus serial device with port "/dev/ttyUSB0" and i have next message in log:

2022-11-24T19:48:53.937Z [ERR]  'gas1' connect failed! Error: Could not locate the bindings file. Tried:
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/build/bindings.node
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/build/Debug/bindings.node
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/build/Release/bindings.node
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/out/Debug/bindings.node
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/Debug/bindings.node
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/out/Release/bindings.node
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/Release/bindings.node
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/build/default/bindings.node
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/compiled/12.22.12/linux/x64/bindings.node
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /usr/lib/node_modules/@frangoteam/fuxa/_pkg/@serialport/bindings/addon-build/default/install-root/bindings.node
 
what i do wrong?

unocelli, 2022-11-25:
Hi, try to install it manually [https://github.com/frangoteam/FUXA/issues/524](https://github.com/frangoteam/FUXA/issues/524)

2345lug, 2022-11-25:
Thank you very much! I remove plugin and install modbus-serial version 8.0.6 with -g key, and serial work now! For expierement, i try to install 8.0.5 and have this: 
gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/lib/node_modules/modbus-serial/node_modules/@serialport/bindings/.node-gyp'
gyp ERR! System Linux 5.15.0-53-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/lib/node_modules/modbus-serial/node_modules/@serialport/bindings
gyp ERR! node -v v12.22.12
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.2.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@9.2.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

unocelli, 2023-02-05:
In master branch you find a version where the library (version 8.0.9) will be installed as standard.

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli