frangoteam/FUXA #611: No historical data displayed on the graph.

haitan981, 2022-11-25:
I don't see historical data showing up on the graph. Everything is empty. How to enable data storage
Thank you so much.
[image]
[image]

unocelli, 2022-11-25:
Hi, you have to enable the tags for history
[image]

haitan981, 2022-11-28:
> Hi, you have to enable the tags for history [image]

thank you so much. I done

unocelli, 2023-02-05:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2023-02-05 by unocelli