frangoteam/FUXA #614: Add Duplicated Tags into  Web API Device

GitTurboy, 2022-11-28:
I created a WebAPI Device which return json data like below:
    {"TimeSeriesDataPoints":[{"HistorianID":579,"Time":"2022-11-28 
    01:57:52.036","Value":10013.09,"Quality":29,"PointName":"DLX13_L80L_D1"},{"HistorianID":580,"Time":"2022-11-28 
   01:57:52.075","Value":224.53,"Quality":29,"PointName":"DLX13_L80L_D101"}]} 

and add tags into this device by choose [PointName] as ArrayID and [Value] as Value field. At this point, all things gone well.

But when I modify the webapi and open  the divice then execute add tags operartion, all tags added again. there will be two tag with same arrayID.

GitTurboy, 2022-12-01:
I found at line 281 of   client\src\app\device\tag-property\tag-property.component.ts add codes like below can resolve this problem
               for(let tag in this.data.device.tags)
                    { if(this.data.device.tags[tag].name==objNode.path)
                        {
                            objNode.enabled =false;
                            break;
                        }
                    }

unocelli, 2023-03-12:
Hi, sorry for the delay.
I was able to reproduce the issue and I will fix it. Thanks for the report.

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli