frangoteam/FUXA #615: Can we make the chart component show historian and realtime  together?

GitTurboy, 2022-11-28:
I means when the view inited, the component can load historian data and show them. after then, new data can be added into the chart and show as a realtime chart.

and by the way, can we add more time range into the historian chart? such as last 1 hour, last 3 hours. or can let the user define their own range

unocelli, 2022-12-01:
Hi,  I don't understand what you mean, automatically refresh the chart in history mode #558 ?

GitTurboy, 2022-12-02:
> Hi, I don't understand what you mean, automatically refresh the chart in history mode #558 ?

yes，when I choose the range as  "last 8 hour", I can see the last 8 hour data end at now, not the time I clikc refresh button.

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli