frangoteam/FUXA #622: Modbus registers bulk import

afbenevides, 2022-12-01:
Im trying to create a file to import all modbus registers in bulk. I downloaded a template file from server, then tried to modify it to add all registers I needed and then re-import it back, but I always get error.

Is there a way to make that? Or is there a location where I could have more info on the import error?

I think it is maybe due to my new values not having valid memory fields? I let these blank.

Does one of you have some kind of guide of how to import like that? My final goal is to automate the file creation of the modbus register list to ease process of modbus setup by importing it directly from scratch.

thanks

unocelli, 2022-12-09:
Hi,
currently there is no documentation for the import function, if you send me the file I can try to see where the problem is.

afbenevides, 2022-12-11:
So I tried it with the json format and it worked with custom files like 2 coils, 2 its, 2hreg and 2 Greg named & numbered by hand. So I can work with that.  

I exported in json and imported after modifications in json also.

Thanks

closed 2022-12-29 by unocelli