frangoteam/FUXA #625: Error when trying to open a FUXA project

IonuDragomir1, 2022-12-06:
Hello,

When I want to open a project, I get in the right corner the message "Project save failed". Even though the project is loaded, I'm getting "TypeError: Cannot read property '0' of undefined" error on console when I want to open a project (please see the below image). 
I'm running FUXA version 1.1.9. 

[image]

How could I please solve this issue?

unocelli, 2022-12-14:
Hi,
happens for any project? also happens when you create a project?

IonuDragomir1, 2022-12-15:
I encountered this error only with some projects. When I create a new project, everything is OK.

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli