frangoteam/FUXA #629: Is it possible to package fuxa in single exe file on windows by pkg tool?

cracy3m, 2022-12-07:
Is it possible to package fuxa in single exe file on Windows platform by pkg tool?

unocelli, 2022-12-14:
interesting feature, should be tried.
originally there was a version with electron but then it became too complicated to maintain compatibility.

chensaki, 2022-12-23:
you can  use "npx pkg" to make single executable file of various platform.
for example, "npx pkg ./main.js -c package.json -t node14-win-x64" for windows.

cracy3m, 2022-12-23:
> you can use "npx pkg" to make single executable file of various platform. for example, "npx pkg ./main.js -c package.json -t node14-win-x64" for windows.

It can be package by the command, but the executable file run with error. 
Like this:
pkg/prelude/bootstrap.js:1876
      throw error;
Error: Filefuxa\node_modules\@foliojs-fork\linebreak\src\classes.trie' was not included into executable at compilation stage. Please recompile adding it as asset or script.
    at error_ENOENT (pkg/prelude/bootstrap.js:657:19)
    at readFileFromSnapshot (pkg/prelude/bootstrap.js:1049:29)
    at Object.readFileSync (pkg/prelude/bootstrap.js:1097:18)
    at Object.<anonymous> (C:\snapshot\node_modules\@frangoteam\fuxa\node_modules\@foliojs-fork\linebreak\src\linebreaker.js:7:36)
    at Module._compile (pkg/prelude/bootstrap.js:1930:22)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at Module.require (pkg/prelude/bootstrap.js:1855:31)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (C:\snapshot\node_modules\@frangoteam\fuxa\node_modules\@foliojs-fork\pdfkit\js\pdfkit.js:9:19)
    at Module._compile (pkg/prelude/bootstrap.js:1930:22)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14) {
  errno: -4058,
  code: 'ENOENT',
  path: 'C:\\snapshot\\node_modules\\@frangoteam\\fuxa\\node_modules\\@foliojs-fork\\linebreak\\src/classes.trie',
  pkg: true
}

chensaki, 2022-12-23:
Please add these package as compiling tips,
for example:
"@foliojs-fork/fontkit": "^1.9.1",
    "@foliojs-fork/linebreak": "^1.1.1",

cracy3m, 2022-12-23:
> Please add these package as compiling tips, for example: "@foliojs-fork/fontkit": "^1.9.1", "@foliojs-fork/linebreak": "^1.1.1",

Pop another issue:
Error: Cannot mkdir in a snapshot. Try mountpoints instead.
    at mkdirFailInSnapshot (pkg/prelude/bootstrap.js:1636:7)
...

 Do you have any examples of success?

chensaki, 2022-12-26:
I can make a single file for running on many platforms. You need to change your operation directory to the current operation directory in the code. "var rootDir = process.cwd();//__dirname;"

To make your own things, you need to understand the source code

cracy3m, 2022-12-26:
The problem is that I'm not familiar with node.js or Angular, so I don't know which paths can be changed

unocelli, 2024-06-23:
You can now use the Electron feature

unocelli, 2024-08-23:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-08-23 by unocelli

cracy3m, 2024-08-25:
Thank you for resolving this issue.

yu-1949, 2024-09-02:
用 Electron 打包成exe启动后。s7连不上是什么问题 ,其他协议正常。日志报这个错 try to create s7  but plugin is missing!
[image]