frangoteam/FUXA #630: Error: array exceeds max variables (20) that can be transferred with ReadMultiVars()

IonuDragomir1, 2022-12-07:
Hello,

When I'm trying to establish a connection to a Vipa M13C PLC, I'm getting " __readVars error! TypeError: Array exceeds max variables (20) that can be transferred with ReadMultiVars()_" error message (please see below image). This error occurs when I add 21st tag and no value is displayed anymore until only 20 tags remain.
Could you please tell me why I get this error and how could it be solved?

[image]

unocelli, 2022-12-14:
Hi,
let me know what version of nodejs and npm you have installed.

IonuDragomir1, 2022-12-15:
Hi,

nodejs version: v14.17.4
npm: 6.14.14

unocelli, 2023-02-05:
This issue should be fixed in the latest version.

IonuDragomir1, 2023-02-16:
Hi,
I have tested on the last version and is not fixed.

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli