frangoteam/FUXA #647: Cann't setting logo

jinyistudio534, 2022-12-21:
My fxua is **1.1.10**. Install in **windows 10** and Node.js is v**16.17.1**.
**I cann't select logo**
[image]

unocelli, 2022-12-26:
Please look solved issues #543

unocelli, 2022-12-29:
you have to search where npm have installed fuxa.
for example ubuntu in windows npm install in `c:\Users\[...]\AppData\Local\Packages\CanonicalGroupLimited.Ubuntu20.04onWindows_79rhkp1fndgsc\LocalState\rootfs\home\[...]\.nvm\versions\node\v14.19.1\lib\node_modules\@frangoteam\fuxa`

unocelli, 2023-02-05:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2023-02-05 by unocelli