frangoteam/FUXA #648: View Editor - Alignment tools needed

concept10, 2022-12-22:
For view development purposes, it would be nice to have exposed alignment tools that distributes view elements (horizontally/vertically) evenly across the canvas.  

Inkscape screenshot (alignment tools)

[image]

unocelli, 2022-12-26:
there are 
[image]

concept10, 2022-12-27:
Thanks for commenting.  I'm sure I have worked with those tools before, but for some reason (possible older version) they were not showing up!   Closing issue!

closed 2022-12-27 by concept10

aniston, 2023-08-17:
[image]

@unocelli in my Docker version 1.1.15-1269 of Fuxa the top Alignment tools are not showing up, the right side Align properties does show, any tip how to show them ?

unocelli, 2023-08-18:
Hi, you have to select more than one element.

aniston, 2023-08-18:
Oh, forgot about that, indeed it shows, Thanks, was concentrated on looking only at the right side align menu.

on a side note, i did select multiple elements before but did not realise that the "Selected Elements" alignment options would show up at the top arther than at the richt side "Alignment properties", hence i missed noticing the top "Alignment menu" and jumped onto this issue.

OffTopic: the MQTT addition is lovely addition, it opens up many possibilities, nice work there.