frangoteam/FUXA #650: The possibility of including the "post" method in the webapi plugin

JefersonBigheti, 2022-12-26:
First of all congratulations for the excellent project. I would like to know if there is a possibility to include the "post" method in the webapi plugin.

unocelli, 2022-12-26:
Hi, thank you for your appreciation.
we have the feature planned but our resources are currently limited.

tayfunonbasioglu, 2022-12-26:
The post method or a webhook-like method can make a lot of work easier. I will eagerly await this.

tayfunonbasioglu, 2022-12-26:
[image]

JefersonBigheti, 2022-12-28:
I'm having trouble implementing it. Would you have an example of how to implement post using your suggestion "environment variable DEVICES"

unocelli, 2022-12-28:
- your web application (device) have to serve the REST API get and post (exp. GET -> http://localhost:8080/api/tags POST -> http://localhost:8080/api/tags)
- you modify the environment parameter [envParams.js](https://github.com/frangoteam/FUXA/blob/master/server/envParams.js) configs.getTags and configs.postTags with your GET/POST url.
- you start fuxa server with arguments `node main.js --env=test`
- in UI you can ask in the connection section the tags configured in your web application through GET  (normally will be asked by start)
[image]

The tags must be in an array (flat structure) as you can see in the following example

```
var tags = [
    {
        id: 'tempA',     // UUID
        name: 'tempA',
        value: 0,
        type: 'number'  // number|boolean|string,`
    },
    {
        id: 'tempB',    // UUID
        name: 'tempB',
        value: 0,
        type: 'number'  // number|boolean|string,`
    }
]
apiApp.get('/api/tags', function (req, res) {       
    res.json(tags);
});

apiApp.post('/api/tags', function (req, res) {
    if (req.body && Array.isArray(req.body)) {
        for(let i = 0; i < req.body.length; i++) {
            for (let y = 0; y < tags.length; y++) {
                if (req.body[i].id === tags[y].id) {
                    tags[y].value = parseFloat(req.body[i].value);
                    y = tags.length;
                }
            }
        }
    }
    res.end();
});
```

JefersonBigheti, 2022-12-29:
The example shown worked very well.
Thanks a lot for the help!

closed 2022-12-29 by JefersonBigheti

RajatDas2020, 2024-08-02:
Hi actually I have a opcua client running in fuxa. From a standalone opcua server I am able to get the opcua variables in fuxa. Now these opcua variables I want to expose to the application layer (upper layer) so that it application can read write the opcua variables. can you kindly guide me how to do that ?

unocelli, 2024-08-17:
Hi, I will use mqtt, install a broker (mosquitto) and export the opcua values

henjoe, 2024-09-27:
Hi @unocelli  I am trying to use this application but seems like the WebAPI properties only works on GET ? 
[image]

Am I missing something here? or the Post method already removed?

children1987, 2025-07-11:
> > Hi, what is the endpoint `http://192.168.56.103:1880/api/fuxa/post` result?
> 
> Yes, That is my post endpoint

@giang321 你好，unocelli 在问你这个url返回的是什么。你可能需要参考下他前面给出的样例

giang321, 2025-07-11:
> > > Hi, what is the endpoint `http://192.168.56.103:1880/api/fuxa/post` result?
> > 
> > 
> > Yes, That is my post endpoint
> 
> [@giang321](https://bgithub.xyz/giang321) 你好，unocelli 在问你这个url返回的是什么。你可能需要参考下他前面给出的样例

是的，非常感谢提醒，我误解了这段话的含义。

giang321, 2025-07-11:
> Hi, what is the endpoint `http://192.168.56.103:1880/api/fuxa/post` result?

Actually, I think Fuxa's POST request might not be reaching the endpoint at all. Could there be any configuration I might have missed?

For the GET method endpoint, I used Node-RED's "http in" node (set to GET method) and wrote the corresponding sample code in the function node of node-red. The GET request from Fuxa responds correctly. The debug node of node-red is shown below and I can get these tag value in fuxa.
[
    {
        "id": "tempA",
        "name": "tempA",
        "type": "number",
        "value": 21
    },
    {
        "id": "tempB",
        "name": "tempB",
        "type": "number",
        "value": 0
    }
]

However, for the POST method, I configured the corresponding URL in node-red "http in" node and set to POST method, but when I check the debug, no information is printed. Does this mean the POST request from Fuxa is not being received? There must be something I haven't set up or understood correctly. In FUXA, where can I configure the payload for POST requests? Could you provide further assistance? Thanks a lot.

unocelli, 2025-08-14:
Hi, follow these steps:
1. define envParams.js
`process.env.DEVICES = `[
    {
        "id":"webapi-A",
        "name":"wApi",
        "type":"WebAPI",
        "configs": {
            "getTags":"http://localhost:1888/api/tags",
            "postTags":"http://localhost:1888/api/tags",
            "requestTimeoutMs":5000,
            "requestIntervalMs":500
        }
    }
]`
;`
2. unpack, install and start node-webapi-server.zip
3. start FUXA server with env param `node FUXA/server/main.js --env=test`
4. go to WebAPI property and press Load Tags
[image]

[node-webapi-server.zip](https://github.com/user-attachments/files/21780559/node-webapi-server.zip)