frangoteam/FUXA #652: Bacnet cannot see objects

aj020, 2022-12-27:
when i try to see object list from bacnet it just shows error
2022-12-27T06:26:49.774Z [ERR]  'bac1' _readObjectList error! Error: BacnetAbort - Reason:4
2022-12-27T06:26:51.221Z [WAR]  'bac1' working (connection || polling) overload!
2022-12-27T06:26:53.229Z [WAR]  'bac1' working (connection || polling) overload!

bencahill, 2023-02-04:
I also couldn't get BACnet devices to populate the list of devices. I found that it worked fine for a BACnet simulator with just a few objects. I found that if I increased the adpuTimeout parameter in index.js, it worked. It seemed like a higher number was required for a higher number of objects. I had to increase it to 60000 (60 seconds) to get it to work on one of my devices with ~75 objects.

aj020, 2023-02-10:
still Same error 'bac1' working (connection || polling) overload! 
also increased the adpuTimeout time i am using bacnet Room stimulator yabe.
still not able to load the objects.