frangoteam/FUXA #653: ROS/ROS2 Support

liushuya7, 2023-01-02:
Hi,

Is it possible to add ROS/ROS2 support?
I know there are libraries like [roslibjs](https://github.com/RobotWebTools/roslibjs) and [rclnodejs](https://github.com/RobotWebTools/rclnodejs).
I think FUXA can become very popular among ROS communities.

liushuya7, 2023-01-16:
I come across a communication protocol called [Zenoh](https://zenoh.io/) which provides bridges to ROS (DDS), MQTT, REST API etc. Is it possible to add a plugin in Fuxa to support Zenoh?

liushuya7, 2023-01-18:
Since Fuxa already supports MQTT, I suppose [mqtt_bridge](http://wiki.ros.org/mqtt_bridge) may work.

unocelli, 2023-01-18:
Hi, interesting and worth a try. let us know if it works.

liushuya7, 2023-04-20:
I'm currently using [RTI OPC UA/DDS Gateway](https://github.com/rticommunity/rticonnextdds-gateway-opcua) to bridge DDS ROS2 to OPC UA. Then I connect OPC UA to FUXA. It works great.