frangoteam/FUXA #657: A development and suggestion for the WEBHOOK

tayfunonbasioglu, 2023-01-08:
Hello Umberto,

First of all, congratulations for the successful work and development. We are looking forward to the new version.

Do you plan to make a plugin about **WebHook** method, which is one of the current issues, where we can send asynchronous data with Json format?

 Sending success and love.

[image]

concept10, 2023-01-08:
Device driver for WebHook? Interesting use case, but interested.  However, why not support the open issue?

tayfunonbasioglu, 2023-01-08:
> Device driver for WebHook? Interesting use case, but interested. However, why not support the open issue?

I didn't see there was an open issue about webhook