frangoteam/FUXA #659: Driven by opc-ua, when the data type is Boolean, when the value is true, the data display error.

hxjackcn, 2023-01-09:
[image]

There is a problem with the latest version, the old version is fine

hxjackcn, 2023-01-09:
Driven by opc-ua, some analogs also have problems.

unocelli, 2023-01-10:
Hi,
We have tested with more than 1000 tags without noting any problems. drivers seem to work very well. try to reinstall node_modules (remove before)

hxjackcn, 2023-01-14:
Already reinstalled node_modules, problem persists

[image]

[image]

unocelli, 2023-01-15:
Hi,
I think we have found and solved the issue

unocelli, 2023-02-05:
This issue should be fixed in the latest version. let me know if you have any issues.

closed 2023-02-05 by unocelli