frangoteam/FUXA #662: specific tag payload - timestamp -  not processed correctly anymore in MQTT

iw65ihg, 2023-01-17:
I experienced a change going from 1.1.9 to 1.1.10
 
the payload "01/16/2023, 19:33:33" was correctly process in 1.1.9.
Same payload is received as "1" in 1.1.10

[image]
[image]

unocelli, 2023-01-17:
Thanks for the report. We have to fix the function that check if is a number.  
with date `01/16/2023` fails
```
    var num = parseFloat(n);
    return !isNaN(num) && isFinite(num);
```

unocelli, 2023-02-05:
This issue should be fixed in the latest version. let me know if you have any issues.

closed 2023-02-05 by unocelli