frangoteam/FUXA #666: How to running new update fuxa in docker container without losing data

zuldemort24, 2023-01-22:
Hi, i am running fuxa on docker using image that i build from Dockerfile in fuxa repository  with this script : docker run -d -p 4200:1881 -v fuxa_appdata2:/usr/src/app/FUXA/server/_appdata -v fuxa_db2:/usr/src/app/FUXA/server/_db -v fuxa_logs2:/usr/src/app/FUXA/server/_logs  myFuxaImage . 

Now i want to rebuild docker image with new version of fuxa on master channel without losing my db data . How to do that ? , i try rebuild image docker and running script above : docker run -d -p 4200:1881 -v fuxa_appdata2:/usr/src/app/FUXA/server/_appdata -v fuxa_db2:/usr/src/app/FUXA/server/_db -v fuxa_logs2:/usr/src/app/FUXA/server/_logs  myNewImage. The data cannot be loading into history . Please help me. Thanks

unocelli, 2023-01-28:
Hi,
you project are saved into _appdata, your daq history are saved into _db, the both are in external volumes, if you install the new docker version you should not lose the data

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli