frangoteam/FUXA #667: Bacnet Device connection error / timeout

Obaq-web, 2023-01-24:
Hi there,

I try to connect to a device which I can see witch "Baceye"-Software from my machine. I know that port 47808 is occupied by another software when I start FUXA. I still get a connection error. What can I look into to get the connection running?

[image]

unocelli, 2023-01-28:
Hi,
there are some settings that we can improve, is currently disabled (commented)
you can modify the line 311 in `fuxa\server\runtime\devices\bacnet\index.js` with:

` client = new bacnet({ port: port });`

and restart the server

there are other settings that we can try to improve but we can not test it because we don't use bacnet

`apduTimeout: 3000`

Obaq-web, 2023-01-29:
Hi,

when I specify the IP on which fuxa is listening I can successfully connect to my bacnet-device. The IP is usually the same IP fuxa is running on. I changed the following line in `fuxa\server\runtime\devices\bacnet\index.js`:

`client = new bacnet();`

to:

```
            client = new bacnet({
                port: 47808,                          // Use BAC1 as communication port
                interface: '192.168.66.212',          // Listen on a specific interface
                broadcastAddress: '192.168.66.255',  // Use the subnet broadcast address
                apduTimeout: 6000
```
As I said I can connect to the bacnet-device that is running on 192.168.66.143.

When I add a bacnet-device with the default code (`client = new bacnet();`) fuxa starts listening on the IP 0.0.0.0. At least that's what my taskmanager is telling me.

Maybe you can change the default code so that the bacnet-client starts on the same IP as fuxa is running on.